mirte-robot/mirte-ros-packages #30: Add a stop_all_motors service

mklomp, 2023-10-18:
While developing, it is quite useful to stop all motors with one command (and optionally, also disable the servo's). This service can then also be used in the python API and web interface to stop all motors from spinning.

ArendJan, 2023-10-19:
That would be nice indeed!

Would this be a 'stop any moving motor right now' service or a 'disable all motors until enable again' service?
Or even both?

mklomp, 2023-10-23:
My first reaction would be to have a 'stop all motors right now' behaviour. I think otherwise we potentially introduce 'bugs' where motors are not responding while they are just disabled.

ArendJan, 2023-10-23:
After discussion:
- add in telemetrix node a service if there is any motor
- on service call, send motorspeed =0% to all motors