mirte-robot/mirte-ros-packages #83: Check topics frequencies and QoS

Rezenders, 2025-08-25:
Hey @ArendJan @SuperJappie08 @mklomp,

I'm trying to find ways to reduce the Mirte Master CPU usage. I noticed that what uses most CPU is the telemetrix node, so I would like to open/follow/be part of the discussion on how to improve it. That is why I'm creating this issue (this is rather a discussion than an issue).

What do you think about the following?

Should mirte_base_control subscribed to the encoder topics with this QoS? Shouldn't it be with the [SensorDataQoS](https://docs.ros.org/en/ros2_packages/rolling/api/rclcpp/generated/classrclcpp_1_1SensorDataQoS.html#class-sensordataqos) as specified in [REP 2003](https://www.ros.org/reps/rep-2003.html)?

```
ros2 topic info -v /io/encoder/front_left
Type: mirte_msgs/msg/Encoder

Publisher count: 1

Node name: telemetrix
Node namespace: /io
Topic type: mirte_msgs/msg/Encoder
Endpoint type: PUBLISHER
GID: 01.0f.e3.1e.96.0b.d3.df.00.00.00.00.00.00.24.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: mirte_base_control
Node namespace: /
Topic type: mirte_msgs/msg/Encoder
Endpoint type: SUBSCRIPTION
GID: 01.0f.e3.1e.98.0b.92.56.00.00.00.00.00.00.26.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

Does the encoder data need to published at 40hz? can it be lower?

```
ros2 topic hz /io/encoder/front_left
WARNING: topic [/io/encoder/front_left] does not appear to be published yet
average rate: 36.595
	min: 0.000s max: 0.074s std dev: 0.01767s window: 38
average rate: 38.241
	min: 0.000s max: 0.074s std dev: 0.01557s window: 78
average rate: 39.160
	min: 0.000s max: 0.074s std dev: 0.01395s window: 120
average rate: 39.523
	min: 0.000s max: 0.085s std dev: 0.01451s window: 161
average rate: 39.863
	min: 0.000s max: 0.085s std dev: 0.01389s window: 203
average rate: 40.162
	min: 0.000s max: 0.085s std dev: 0.01390s window: 246
```

Should `ros2_arm_control_hw_interface` subscribe to the servo's topics with the System QoS or Sensor QoS?

```
ros2 topic info -v  /io/servo/hiwonder/elbow/position
Type: mirte_msgs/msg/ServoPosition

Publisher count: 1

Node name: telemetrix
Node namespace: /io
Topic type: mirte_msgs/msg/ServoPosition
Endpoint type: PUBLISHER
GID: 01.0f.e3.1e.96.0b.d3.df.00.00.00.00.00.00.57.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: ros2_arm_control_hw_interface
Node namespace: /
Topic type: mirte_msgs/msg/ServoPosition
Endpoint type: SUBSCRIPTION
GID: 01.0f.e3.1e.98.0b.92.56.00.00.00.00.00.00.51.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

Does the servo positions need to be published at 50hz?

```
ros2 topic hz  /io/servo/hiwonder/elbow/position
WARNING: topic [/io/servo/hiwonder/elbow/position] does not appear to be published yet
average rate: 46.763
	min: 0.000s max: 0.088s std dev: 0.02637s window: 52
average rate: 47.807
	min: 0.000s max: 0.123s std dev: 0.03070s window: 105
average rate: 51.197
	min: 0.000s max: 0.123s std dev: 0.02870s window: 164
average rate: 50.355
	min: 0.000s max: 0.123s std dev: 0.02765s window: 213
average rate: 50.920
	min: 0.000s max: 0.123s std dev: 0.02697s window: 267
average rate: 50.860
	min: 0.000s max: 0.123s std dev: 0.02629s window: 319
average rate: 51.040
	min: 0.000s max: 0.123s std dev: 0.02590s window: 373
```

Rezenders, 2025-08-25:
I asked codex (from chatgpt) if it could find possible improvements. This is the output: https://chatgpt.com/s/cd_68ac869572a48191bd3b94b0417ca2e6

ArendJan, 2025-08-25:
Chats #3 I have in my pr, but needs a check

SuperJappie08, 2025-08-25:
Hi @Rezenders, the subscribers you mention are actually subscribing via SensorDataQos (or something equivalent).
They are published as SystemDefaultQoS as specified in REP2000, which improves compatibility (since publishing as SensorDataQoS is not advised, since it can lead to QoS incompatibilities).

I'm currently in the process of redoing the (base) control hardware interfaces, so those will definitely be looked at.

The encoders should preferably even be published at a higher frequency, so in my work I currently have a separate update topic with just the updates.

The arm servos I'm unsure about, since we can practically only control them at a lower frequency. So I would imagine that needs to be tuned according to those limitations.

My response to the ChatGPT suggestions:
1. The `MultiThreadedExecutor` mostly helps us here, since it allows for multiple _devices_ to be used at once from ROS, which allows non-colliding/controlling services to be executed simultaneously, which was a limitation in the original ROS 1 implementation.
2. Most of the logging isn't that bad, as it mostly gets used during initialization and errors. But it should definitely be checked (and all converted to ROS logging for consistency and tunability (since then some levels can be ignored easily, which means not outputting them to a file or `/rosout`)).
3. Making sure each device has its own callback group allows for the optimizations mentioned in my response to point 1. However, this could probably be tuned better. An issue @ArendJan already investigated is each device having their own timers, which most likely is the main performance hit[^1] (combined with the serialization that happens on every publish).
Simply joining all timers (and not publishing on TMX Receive) has its own complications, especially for control purposes.
Overall, I agree that performance gains are achievable here.
4. The mutexes are put in place because of the async communication of TMX and the ROS timer, which caused race conditions where memory was read and written to simultaneously. The messages could be saved and republished with updated fields; then the `get_header` function should not be used. However, I think the message allocation is not the main issue, as the RMW most likely needs to copy the message anyway for its own internal buffer/serialization and QoS behavior.
5. This only happens during initialization, which is probably fine, as I presume it is smart enough to join the thread when done. I put that in place because a lot of communication occurs during initialization, and it caused many issues (due to timeouts) or delays specifically with the HiWonder Servos, but I am currently uncertain about the exact reason anymore, since it was a while ago, but there was some sort of motivation to treat those differently.
6. I agree that the parser may overlog (especially some of the `cout` lines), but this only happens during initialization, and therefore changing this will **not** have any noticeable runtime. Improvements.

The final prompt response mostly mentions code that only gets executed once in the lifetime of the Telemetrix, which means that isn't the place to '_win_' performance.
The middle two points are something that could be considered, but I'm not familiar enough with that section of the code to give a useful opinion without spending a lot of time refamiliarizing myself with it.

[^1]: @ArendJan, I just realized that the issue might be the constant timer rescheduling by calling timer reset after being published in the TMX callback. We should probably investigate that as well (because keeping the timers separate is conceptually simpler than combining them).

ArendJan, 2025-08-26:
I can look into the footnote of @SuperJappie08 , but that will probably be next week. The best optimization is to remove parts from the config that you don't need, like a sonar or the arm servos if you don't use them.

ArendJan, 2025-08-26:
@Rezenders could you try my pull request and see if that is better: https://github.com/mirte-robot/mirte-ros-packages/pull/81
You'll need to update the pico on your robot: 
```
cd /usr/local/src/mirte/mirte-telemetrix4rpipico/
git pull # check that you're on the mirte-robot repo with the develop branch
git submodules update ...
cd /usr/local/src/mirte/mirte-install-scripts/
./run_arduino.sh upload pico # this will compile and upload the new code
```

ArendJan, 2025-08-26:
Nice! Network is indeed a big problem when using the point clouds & camera images. You can always use ethernet if you don't mind the cable